Guilouz/Marlin-SuperRacer-MKS-Nano-V3 #3: Some of your install instructions are not correct

Xiondewah, 2021-08-30:
Hi - love the firmware, but some of your instructions are not right.

The PID commands you have wrong - ie should be 
M301 KP24.03 KI2.22 KD64.88 - same for bed please.

Also, can you add a note on the G33 to say you need to add the bed levelling sensor before running the command?  Otherwise the head will crash into the bed.

Many thanks :-)

Guilouz, 2021-08-30:
My commands are good... M301 PKp IKi DKd and M304 PKp IKi DKd are generic commands where Kp, Ki and Kd is different for all printer (obtained with M303).

Xiondewah, 2021-08-30:
But when I entered your commands on my SR it failed to recognise them.  Only when I put them in as above did it change the settings.  Since this firmware is printer-specific, wouldn't it be better to have the specific instructions?  Beginners might struggle otherwise.

Guilouz, 2021-08-30:
You must replace Kp, Ki and Kd with value obtained with M303 command.

Xiondewah, 2021-08-30:
OK I understand now.  Perhaps you could just clarify this a little?
This is not intended as a criticism.  I have written numerous user and software manuals professionally, and understand that people usually read these things and take them quite literally.
Thanks.

closed 2021-08-31 by Guilouz