Guilouz/Marlin-SuperRacer-MKS-Nano-V3 #5: 4096 not an allocation size for formatting on Windows 10

dominickp, 2021-08-31:
Is it OK to choose something other than 4096? Or should Windows users use another application to format their SD cards?

[image]

Guilouz, 2021-08-31:
Depending of your SD Card, mine have 4096. You can choose default.

[image]

[image]

dominickp, 2021-08-31:
Thanks. I went with the default.

closed 2021-08-31 by dominickp