Guilouz/Marlin-SuperRacer-MKS-Nano-V3 #7: Print crash from Cura generated Gcode

aifolfr, 2021-09-15:
Hi,

I have a systematic crash when I print a TempTower generated from Cura/Extensions/parts for Calibration and the associated code from "Modify gcode/Tempfan tower".
The fan function is not selected.
After 8 to 12 layers, the head stop at the left tower, no message on screen. I cancel the print from the screen, I have the message associated with it and from there the Printer is stuck, the message stays on screen forever and I must shut off the printer.
The Temptower works with my other printers.

aifolfr, 2021-09-19:
Frozen again today with an every day print, but I was playing quite a lot with temperatures on screen so til today crashes happened with temperature changes while printing.

closed 2021-12-22 by Guilouz

mithrandir63, 2022-01-13:
Bonjour,
J'ai le même problème.... Une impression sur 2 plante.... Avec le même Gcodes.... J'ai changé la carte sd.... Reformaté fat32.... J'ai suivi toute les étapes des mise à jour... Avec les reset eeprom... .. toujours après quelques minutes de la 2 ème couche...
Je suis sûre cura 4.12.1...
Avez vous une idée ?????.
J'aurais bien aimé resté sur ce firmware...
Cdlt.