Guilouz/Marlin-SuperRacer-MKS-Nano-V3 #8: Build

ethanfel, 2021-11-16:
Hello :)

thanks for the firmware. I've been using a while now.

I've got an issue with my board, there's a short in the HE0 and the hotend overheat until thermal runaway protection is triggered.
My solution is to switch the pins for HE0 and HE1 in marlin (already tested in klipper but i prefer marlin)

https://github.com/ethanfel/Marlin-SuperRacer-MKS-Nano-V3

My issue is that i can't build it with platformio. Tested in Linux and Windows, i don't have an error. it just  fail with a buncch of yellow warnings.

Could you build this version for me ? until i get a new nano v3 i hope from FLsun. Thanks

Guilouz, 2021-11-16:
What version do you use USB or SDCARD ?

ethanfel, 2021-11-16:
SDcard.

Guilouz, 2021-11-16:
Here we go :
[Robin_nano_v3.bin.zip](https://github.com/Guilouz/Marlin-SuperRacer-MKS-Nano-V3/files/7550641/Robin_nano_v3.bin.zip)

ethanfel, 2021-11-16:
> Here we go : [Robin_nano_v3.bin.zip](https://github.com/Guilouz/Marlin-SuperRacer-MKS-Nano-V3/files/7550641/Robin_nano_v3.bin.zip)

thank you very much  !)

Guilouz, 2021-11-17:
Perfect! I'm on MacOS.

closed 2021-11-17 by ethanfel