Guilouz/Marlin-SuperRacer-MKS-Nano-V3 #10: I have finally able to compile and flashed my SR. i wanted to activate buzzer with M114 B1 and save it with M500. but after restart buzzer function gone.

ayesim, 2021-12-17:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

tsviper, 2021-12-29:
Same issue here.  Is there a way to make this save?

ayesim, 2022-01-01:
I think bed leveling mesh size messing EEPROM data structure and not able to save it sometimes effecting language save function.
In Configuration.h file change value 9 to 5 (#define GRID_MAX_POINTS_X 5). it will effect bed leveling points 9x9 to 5x5. i have tried 9x9 with original FLSUN firmware and I've got same buzzer loss and language problem. but when I return back to 5x5 configuration fault was gone. M114 B0 and M114 B1 G code function is already implemented in FLSUN original firmware.  
FLSUN has some buzzer activation function in the MarlinSerial.cpp file and if you add enable_buzzer(); command some suitable location in the code it will activate buzzer function. if you add disable_buzzer(); command in the source it will disable buzzer. buzzer enable flag located in the eeprom but eeprom address location collide with other datas and mess this flag including language.

closed 2022-02-14 by Guilouz