Guilouz/Marlin-SuperRacer-MKS-Nano-V3 #12: Build (linker) fails with tons of undefined symbols

compihu, 2021-12-22:
`[compi@arwen Marlin-SuperRacer-MKS-Nano-V3]$ pio run
Processing mks_robin_nano_v3_usb_flash_drive_msc (platform: ststm32@~10.0; board: genericSTM32F407VGT6; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F407VGT6.html
PLATFORM: ST STM32 (10.0.1) > STM32F407VG (128k RAM, 64k CCM RAM, 1024k Flash
HARDWARE: STM32F407VGT6 168MHz, 128KB RAM, 1MB Flash
DEBUG: Current (jlink) External (jlink, stlink)
PACKAGES: 
 - framework-arduinoststm32 0.0.0 
 - framework-cmsis 2.50501.200527 (5.5.1) 
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 13 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <TMCStepper> 0.7.1
|   |-- <SPI> 1.0
|   |-- <SoftwareSerial> 1.0
|-- <LiquidCrystal> 1.5.0
|   |-- <Wire> 1.0
|-- <SPI> 1.0
|-- <Servo> 1.1.2
|-- <SoftwareSerial> 1.0
|-- <IWatchdog> 1.0.0
|-- <Wire> 1.0
|-- <EEPROM> 2.0.1
|-- <qr_encode> 0.0.0+20201117144502
Building in release mode
Linking .pio/build/mks_robin_nano_v3_usb_flash_drive_msc/firmware.elf
/home/compi/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: .pio/build/mks_robin_nano_v3_usb_flash_drive_msc/src/src/MarlinCore.cpp.o: in function `finishSDPrinting()':
/mnt/userdata/compi/work/ext/Marlin/Marlin-SuperRacer-MKS-Nano-V3/Marlin/src/MarlinCore.cpp:396: undefined reference to `print_thr_adress_string(int, int, char*)'
/home/compi/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: /mnt/userdata/compi/work/ext/Marlin/Marlin-SuperRacer-MKS-Nano-V3/Marlin/src/MarlinCore.cpp:397: undefined reference to `print_thr_adress_string(int, int, char*)'
/home/compi/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: /mnt/userdata/compi/work/ext/Marlin/Marlin-SuperRacer-MKS-Nano-V3/Marlin/src/MarlinCore.cpp:404: undefined reference to `total_time'
/home/compi/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: /mnt/userdata/compi/work/ext/Marlin/Marlin-SuperRacer-MKS-Nano-V3/Marlin/src/MarlinCore.cpp:404: undefined reference to `my_print_state'
/home/compi/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: .pio/build/mks_robin_nano_v3_usb_flash_drive_msc/src/src/MarlinCore.cpp.o: in function `setFLSunHours(unsigned long)':
/mnt/userdata/compi/work/ext/Marlin/Marlin-SuperRacer-MKS-Nano-V3/Marlin/src/MarlinCore.cpp:1304: undefined reference to `print_thr_adress_string(int, int, char*)'
/home/compi/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: .pio/build/mks_robin_nano_v3_usb_flash_drive_msc/src/src/MarlinCore.cpp.o: in function `setFLSunLanguage(unsigned short)':
/mnt/userdata/compi/work/ext/Marlin/Marlin-SuperRacer-MKS-Nano-V3/Marlin/src/MarlinCore.cpp:1310: undefined reference to `change_en()'
/home/compi/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: /mnt/userdata/compi/work/ext/Marlin/Marlin-SuperRacer-MKS-Nano-V3/Marlin/src/MarlinCore.cpp:1314: undefined reference to `change_zh_CN()'
/home/compi/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: /mnt/userdata/compi/work/ext/Marlin/Marlin-SuperRacer-MKS-Nano-V3/Marlin/src/MarlinCore.cpp:1318: undefined reference to `change_zh_TW()'
/home/compi/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: /mnt/userdata/compi/work/ext/Marlin/Marlin-SuperRacer-MKS-Nano-V3/Marlin/src/MarlinCore.cpp:1322: undefined reference to `change_ru()'
/home/compi/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: /mnt/userdata/compi/work/ext/Marlin/Marlin-SuperRacer-MKS-Nano-V3/Marlin/src/MarlinCore.cpp:1326: undefined reference to `change_fr()'
/home/compi/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: /mnt/userdata/compi/work/ext/Marlin/Marlin-SuperRacer-MKS-Nano-V3/Marlin/src/MarlinCore.cpp:1330: undefined reference to `change_es()'
/home/compi/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: /mnt/userdata/compi/work/ext/Marlin/Marlin-SuperRacer-MKS-Nano-V3/Marlin/src/MarlinCore.cpp:1334: undefined reference to `change_de()'
/home/compi/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: /mnt/userdata/compi/work/ext/Marlin/Marlin-SuperRacer-MKS-Nano-V3/Marlin/src/MarlinCore.cpp:1338: undefined reference to `change_jp()'
/home/compi/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: .pio/build/mks_robin_nano_v3_usb_flash_drive_msc/src/src/MarlinCore.cpp.o: in function `abortSDPrinting()':
/mnt/userdata/compi/work/ext/Marlin/Marlin-SuperRacer-MKS-Nano-V3/Marlin/src/MarlinCore.cpp:386: undefined reference to `jump_to(int)'
/home/compi/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: .pio/build/mks_robin_nano_v3_usb_flash_drive_msc/src/src/MarlinCore.cpp.o: in function `loop':
/mnt/userdata/compi/work/ext/Marlin/Marlin-SuperRacer-MKS-Nano-V3/Marlin/src/MarlinCore.cpp:1371: undefined reference to `jump_to(int)'
/home/compi/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: /mnt/userdata/compi/work/ext/Marlin/Marlin-SuperRacer-MKS-Nano-V3/Marlin/src/MarlinCore.cpp:1379: undefined reference to `tmc_jump'
/home/compi/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: .pio/build/mks_robin_nano_v3_usb_flash_drive_msc/src/src/MarlinCore.cpp.o: in function `MarlinSerial::begin(unsigned long)':
/mnt/userdata/compi/work/ext/Marlin/Marlin-SuperRacer-MKS-Nano-V3/Marlin/src/inc/../HAL/./STM32/MarlinSerial.h:37: undefined reference to `MarlinSerial::begin(unsigned long, unsigned char)'
/home/compi/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: .pio/build/mks_robin_nano_v3_usb_flash_drive_msc/src/src/MarlinCore.cpp.o: in function `setup':
/mnt/userdata/compi/work/ext/Marlin/Marlin-SuperRacer-MKS-Nano-V3/Marlin/src/MarlinCore.cpp:1279: undefined reference to `print_thr_adress_string(int, int, char*)'
/home/compi/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: /mnt/userdata/compi/work/ext/Marlin/Marlin-SuperRacer-MKS-Nano-V3/Marlin/src/MarlinCore.cpp:1280: undefined reference to `print_thr_adress_string(int, int, char*)'
/home/compi/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: /mnt/userdata/compi/work/ext/Marlin/Marlin-SuperRacer-MKS-Nano-V3/Marlin/src/MarlinCore.cpp:1283: undefined reference to `jump_to(int)'
/home/compi/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: /mnt/userdata/compi/work/ext/Marlin/Marlin-SuperRacer-MKS-Nano-V3/Marlin/src/MarlinCore.cpp:1293: undefined reference to `enable_buzzer()'
/home/compi/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: /mnt/userdata/compi/work/ext/Marlin/Marlin-SuperRacer-MKS-Nano-V3/Marlin/src/MarlinCore.cpp:1288: undefined reference to `jump_to(int)'
/home/compi/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: /mnt/userdata/compi/work/ext/Marlin/Marlin-SuperRacer-MKS-Nano-V3/Marlin/src/MarlinCore.cpp:1297: undefined reference to `disable_buzzer()'
[... cut ...]
================================================================================ [FAILED] Took 15.13 seconds ================================================================================

Environment                            Status    Duration
-------------------------------------  --------  ------------
mks_robin_nano_v3_usb_flash_drive_msc  FAILED    00:00:15.130
=========================================================================== 1 failed, 0 succeeded in 00:00:15.130 ============`

ayesim, 2021-12-22:
my understanding, FLSUN give us source files but they did not publish MarlinSerial.cpp file which is contain these missing symbols FLSUN LCD related communication definitions. that's why you have to download official source file and find MarlinSerial.CPP.o and below is from readme.txt
   
The first compilation will fail
then,copy MarlinSerial.cpp.o file to "NanoV3-TFcard\.pio\build\mks_robin_nano_v3_usb_flash_drive_msc\src\src\HAL\STM32"
then ,compile again ,it will succeed.

Guilouz, 2021-12-22:
Read this : https://github.com/Guilouz/Marlin-SuperRacer-MKS-Nano-V3/tree/main/_README

compihu, 2021-12-22:
That is very clearly a violation of Marlin's license.

Guilouz, 2021-12-22:
Nope, Marlin is under GPL licence. 
The purpose of the GNU GPL license, according to its creators, is to guarantee the user the following rights (called freedoms) over a computer program:

Freedom 0. The freedom to run the software, for any purpose;
Freedom 1. The freedom to study the operation of a program and of the adapter to its needs, which requires access to source codes;
Freedom 2. The freedom to redistribute copies;
Freedom 3. The obligation to provide the community with the benefit of the modified versions.

compihu, 2021-12-22:
GPL clearly states that you have to provide ALL the source code required to rebuild the binary. No escape routes. If FLSun sells the product with a binary build from a GPL-covered source the **MUST** make **ALL** the source code required to reproduce that binary available.
They are violating the license also by distributing the MarlinSerial.o which is derived from the GPL licensed MarlinSerial.cpp from Marlin. Even the derived work is subject to the GPL (GPL is a so-called viral license)

Guilouz, 2021-12-22:
Go tell them ... It's already good that they gave the sources because some brands do not give them. And again they published them because I asked them, insisting on the fact that they were obliged to provide the sources.

closed 2022-01-02 by Guilouz