Guilouz/Marlin-SuperRacer-MKS-Nano-V3 #16: Can't find where to place MarlinSerial.cpp.o file.

KnownOriginator, 2022-02-14:
I'm trying to compile the firmware for my Super Racer specifically to enable linear advance and to use a USB thumb drive in place of a microSD card. I've followed Teaching Tech's youtube guide on using VSCode and the Auto Build Marlin plugin. I've gotten as far as the actual building of the firmware, and I see that I need to place a "MarlinSerial.cpp.o" file in a "\src\HAL\STM32" folder. I however cannot for the life of me find where this folder is. Any help would be much appreciated!

Guilouz, 2022-02-14:
You must enable hidden folders in Windows to show .pio folder.

KnownOriginator, 2022-02-14:
But where is the .pio folder in the first place? I have hidden folders enabled already.

Guilouz, 2022-02-14:
Yes to the root folder. And don't use Auto Build Marlin plugin to compile firmware.

KnownOriginator, 2022-02-14:
What root folder where? I've never done this kind of thing before.

Guilouz, 2022-02-14:
Here
[image]

KnownOriginator, 2022-02-14:
Just that folder within the zip file I downloaded here? I know you said not to use the auto marlin thing but if I do use it, when I get to the step where it's "meant" to fail, it removes almost everything within that folder. Can you link me to a proper tutorial on doing this? It's all very confusing for me.

KnownOriginator, 2022-02-14:
Okay I *think* I've gotten it? There is 2 bin files now, a firmware.bin and a robin_nano_v3.bin. I hope I'm right in saying it's the firmware.bin file?

KnownOriginator, 2022-02-14:
When I first extract the source code I can find exactly the folders I need. However when I open the base folder in vscode (without auto marlin) and edit my config file(s) for some reason it deletes those folders?

KnownOriginator, 2022-02-15:
https://drive.google.com/file/d/1eujrxxKa68cjZ-PCqM81brIXr5hjWJW6/view?usp=sharing

Sorry, best way I could show what's going wrong on my end.

closed 2022-02-16 by KnownOriginator