Guilouz/Marlin-SuperRacer-MKS-Nano-V3 #17: Percentage of print and print time both stuck at 0 before/during/after printing.

KnownOriginator, 2022-02-14:
I am currently using the pre-compiled sdcard version of the firmware (it works amazingly btw, awesome job), but after a while for some reason my timer and percentage thing on the screen stopped working? They both just stay at 0% and 0h0m at all times. Doesn't affect anything, and everything else seems to work just fine. Just an odd thing. Not sure if it even could be fixed tbh.

If it's a known issue just close this issue without a response, I'll understand.

Guilouz, 2022-02-14:
Yes it's a jnow issue, you just need to full format (not a rapid format) your sdcard.

KnownOriginator, 2022-02-14:
Interesting, will report back with my results just as soon as I'm done printing my current print, been going all night and I'm not gonna stop now!

closed 2022-02-15 by KnownOriginator