Guilouz/Marlin-SuperRacer-MKS-Nano-V3 #22: Unable to Compile - Outdated Marlin

antdragone, 2022-03-08:
Unable to Compile

I've tried to compile the code to support the E3D Revo hotend, but I am unable to Compile the code. The Marlin devs are saying that this is likely caused by the Marlin version being outdated.

Really just trying to compile the code to support an E3D Thermistor.

Here are the errors I'm getting when attempting to compile with PlatformIO:

```
E:\ADragone\OneDrive\Computers\Nomad\Desktop\Marlin-SuperRacer-MKS-Nano-V3-26-02-2022> platformio run -e mks_robin_nano_v3   
Processing mks_robin_nano_v3 (platform: ststm32@~10.0; board: genericSTM32F407VGT6; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------Tool Manager: Installing platformio/framework-arduinoststm32 @ ~4.10900.0
Unpacking  [####################################]  100%
Tool Manager: framework-arduinoststm32 @ 4.10900.200819 has been installed!
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F407VGT6.html
PLATFORM: ST STM32 (10.0.1) > STM32F407VG (128k RAM, 64k CCM RAM, 1024k Flash
HARDWARE: STM32F407VGT6 168MHz, 128KB RAM, 1MB Flash
DEBUG: Current (jlink) External (jlink, stlink)
PACKAGES: 
 - framework-arduinoststm32 4.10900.200819 (1.9.0) 
 - framework-cmsis 2.50501.200527 (5.5.1) 
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Library Manager: Installing bitbucket-fmalpartida/LiquidCrystal @ 1.5.0
KeyError: 'framework-stm32cube':
  File "C:\Users\Anthony\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 181:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\Anthony\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Script\SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Anthony\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Script\SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\Anthony\.platformio\platforms\ststm32@10.0.1\builder\main.py", line 99:
    target_elf = env.BuildProgram()
  File "C:\Users\Anthony\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Util.py", line 742:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Anthony\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 62:
    env.ProcessProjectDeps()
  File "C:\Users\Anthony\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Util.py", line 742:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Anthony\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 141:
    project_lib_builder = env.ConfigureProjectLibBuilder()
  File "C:\Users\Anthony\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Util.py", line 742:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Anthony\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 1104:
    project.install_dependencies()
  File "C:\Users\Anthony\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 940:
    lm.install(spec)
  File "C:\Users\Anthony\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 48:
    pkg = self._install(
  File "C:\Users\Anthony\.platformio\penv\lib\site-packages\platformio\package\manager\library.py", line 105:
    if is_builtin_lib(spec.name):
  File "C:\Users\Anthony\.platformio\penv\lib\site-packages\platformio\commands\lib\helpers.py", line 49:
    for storage in storages or get_builtin_libs():
  File "C:\Users\Anthony\.platformio\penv\lib\site-packages\platformio\commands\lib\helpers.py", line 34:
    for storage in p.get_lib_storages():
  File "C:\Users\Anthony\.platformio\penv\lib\site-packages\platformio\platform\base.py", line 214:
    pkg = self.get_package(opts["package"])
  File "C:\Users\Anthony\.platformio\penv\lib\site-packages\platformio\platform\_packages.py", line 30:
    return self.pm.get_package(spec or self.get_package_spec(name))
  File "C:\Users\Anthony\.platformio\penv\lib\site-packages\platformio\platform\_packages.py", line 22:
    owner=self.packages[name].get("owner"),
================================================ [FAILED] Took 59.03 seconds ================================================

Environment        Status    Duration
-----------------  --------  ------------
mks_robin_nano_v3  FAILED    00:00:59.035
=========================================== 1 failed, 0 succeeded in 00:00:59.035 ===========================================
E:\ADragone\OneDrive\Computers\Nomad\Desktop\Marlin-SuperRacer-MKS-Nano-V3-26-02-2022> echo "done" >"C:\Users\Anthony\AppData\Local\Temp\ipc"
E:\ADragone\OneDrive\Computers\Nomad\Desktop\Marlin-SuperRacer-MKS-Nano-V3-26-02-2022> 
```

[Marlin-SuperRacer-MKS-Nano-V3-26-02-2022 (2).zip](https://github.com/Guilouz/Marlin-SuperRacer-MKS-Nano-V3/files/8209856/Marlin-SuperRacer-MKS-Nano-V3-26-02-2022.2.zip)

antdragone, 2022-03-09:
I attempted to compile this outside of VS within python and Platoformio, and I got this again;

```
E:\ADragone\OneDrive\Computers\Nomad\Desktop\Marlin-SuperRacer-MKS-Nano-V3-26-02-2022>"C:\Users\Anthony\.platformio\penv\Scripts\platformio" run -e mks_robin_nano_v3
Processing mks_robin_nano_v3 (platform: ststm32@~10.0; board: genericSTM32F407VGT6; framework: arduino)
------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F407VGT6.html
PLATFORM: ST STM32 (10.0.1) > STM32F407VG (128k RAM, 64k CCM RAM, 1024k Flash
HARDWARE: STM32F407VGT6 168MHz, 128KB RAM, 1MB Flash
DEBUG: Current (jlink) External (jlink, stlink)
PACKAGES:
 - framework-arduinoststm32 4.10900.200819 (1.9.0)
 - framework-cmsis 2.50501.200527 (5.5.1)
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Library Manager: Installing bitbucket-fmalpartida/LiquidCrystal @ 1.5.0
KeyError: 'framework-stm32cube':
  File "C:\Users\Anthony\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 181:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\Anthony\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Script\SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Anthony\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Script\SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\Anthony\.platformio\platforms\ststm32@10.0.1\builder\main.py", line 99:
    target_elf = env.BuildProgram()
  File "C:\Users\Anthony\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Util.py", line 742:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Anthony\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 62:
    env.ProcessProjectDeps()
  File "C:\Users\Anthony\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Util.py", line 742:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Anthony\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 141:
    project_lib_builder = env.ConfigureProjectLibBuilder()
  File "C:\Users\Anthony\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Util.py", line 742:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Anthony\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 1104:
    project.install_dependencies()
  File "C:\Users\Anthony\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 940:
    lm.install(spec)
  File "C:\Users\Anthony\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 48:
    pkg = self._install(
  File "C:\Users\Anthony\.platformio\penv\lib\site-packages\platformio\package\manager\library.py", line 105:
    if is_builtin_lib(spec.name):
  File "C:\Users\Anthony\.platformio\penv\lib\site-packages\platformio\commands\lib\helpers.py", line 49:
    for storage in storages or get_builtin_libs():
  File "C:\Users\Anthony\.platformio\penv\lib\site-packages\platformio\commands\lib\helpers.py", line 34:
    for storage in p.get_lib_storages():
  File "C:\Users\Anthony\.platformio\penv\lib\site-packages\platformio\platform\base.py", line 214:
    pkg = self.get_package(opts["package"])
  File "C:\Users\Anthony\.platformio\penv\lib\site-packages\platformio\platform\_packages.py", line 30:
    return self.pm.get_package(spec or self.get_package_spec(name))
  File "C:\Users\Anthony\.platformio\penv\lib\site-packages\platformio\platform\_packages.py", line 22:
    owner=self.packages[name].get("owner"),
============================================== [FAILED] Took 2.57 seconds ==============================================
Environment        Status    Duration
-----------------  --------  ------------
mks_robin_nano_v3  FAILED    00:00:02.570
======================================== 1 failed, 0 succeeded in 00:00:02.570 ========================================

E:\ADragone\OneDrive\Computers\Nomad\Desktop\Marlin-SuperRacer-MKS-Nano-V3-26-02-2022>
```

antdragone, 2022-03-09:
I may have found a similar issue, attempting the fix by updating platformio
https://github.com/esphome/issues/issues/2289

antdragone, 2022-03-09:
Sadly, that did not work, still getting the same error;
```
E:\ADragone\OneDrive\Computers\Nomad\Desktop\Marlin-SuperRacer-MKS-Nano-V3-26-02-2022>"C:\Users\Anthony\.platformio\penv\Scripts\platformio" run -e mks_robin_nano_v3
Processing mks_robin_nano_v3 (platform: ststm32@~10.0; board: genericSTM32F407VGT6; framework: arduino)
------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F407VGT6.html
PLATFORM: ST STM32 (10.0.1) > STM32F407VG (128k RAM, 64k CCM RAM, 1024k Flash
HARDWARE: STM32F407VGT6 168MHz, 128KB RAM, 1MB Flash
DEBUG: Current (jlink) External (jlink, stlink)
PACKAGES:
 - framework-arduinoststm32 4.10900.200819 (1.9.0)
 - framework-cmsis 2.50501.200527 (5.5.1)
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Library Manager: Installing bitbucket-fmalpartida/LiquidCrystal @ 1.5.0
KeyError: 'framework-stm32cube':
  File "C:\Users\Anthony\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 181:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\Anthony\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Script\SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Anthony\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Script\SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\Anthony\.platformio\platforms\ststm32@10.0.1\builder\main.py", line 99:
    target_elf = env.BuildProgram()
  File "C:\Users\Anthony\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Util.py", line 742:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Anthony\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 62:
    env.ProcessProjectDeps()
  File "C:\Users\Anthony\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Util.py", line 742:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Anthony\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 141:
    project_lib_builder = env.ConfigureProjectLibBuilder()
  File "C:\Users\Anthony\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Util.py", line 742:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Anthony\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 1104:
    project.install_dependencies()
  File "C:\Users\Anthony\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 940:
    lm.install(spec)
  File "C:\Users\Anthony\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 48:
    pkg = self._install(
  File "C:\Users\Anthony\.platformio\penv\lib\site-packages\platformio\package\manager\library.py", line 105:
    if is_builtin_lib(spec.name):
  File "C:\Users\Anthony\.platformio\penv\lib\site-packages\platformio\commands\lib\helpers.py", line 49:
    for storage in storages or get_builtin_libs():
  File "C:\Users\Anthony\.platformio\penv\lib\site-packages\platformio\commands\lib\helpers.py", line 34:
    for storage in p.get_lib_storages():
  File "C:\Users\Anthony\.platformio\penv\lib\site-packages\platformio\platform\base.py", line 214:
    pkg = self.get_package(opts["package"])
  File "C:\Users\Anthony\.platformio\penv\lib\site-packages\platformio\platform\_packages.py", line 30:
    return self.pm.get_package(spec or self.get_package_spec(name))
  File "C:\Users\Anthony\.platformio\penv\lib\site-packages\platformio\platform\_packages.py", line 22:
    owner=self.packages[name].get("owner"),
============================================== [FAILED] Took 3.06 seconds ==============================================

Environment        Status    Duration
-----------------  --------  ------------
mks_robin_nano_v3  FAILED    00:00:03.064
======================================== 1 failed, 0 succeeded in 00:00:03.064 ========================================

E:\ADragone\OneDrive\Computers\Nomad\Desktop\Marlin-SuperRacer-MKS-Nano-V3-26-02-2022>
```

antdragone, 2022-03-09:
I was able to replicate this with FLSun's open source firmware too. Exact same error. I'm genuinely stumped here.

antdragone, 2022-03-09:
Honestly, i'm just trying to apply the changes outlined here;
https://e3d-online.zendesk.com/hc/en-us/articles/4411708912017-Marlin-2-0-firmware-guide-Revo-Hemera-

Guilouz, 2022-03-09:
See readme : https://github.com/Guilouz/Marlin-SuperRacer-MKS-Nano-V3/tree/main/_README

antdragone, 2022-03-09:
I have done that, and I get the same error;

```
E:\ADragone\OneDrive\Computers\Nomad\Desktop\Marlin-SuperRacer-MKS-Nano-V3-26-02-2022> platformio run -e mks_robin_nano_v3_usb_flash_drive
Processing mks_robin_nano_v3_usb_flash_drive (platform: ststm32@~10.0; board: genericSTM32F407VGT6; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F407VGT6.html
PLATFORM: ST STM32 (10.0.1) > STM32F407VG (128k RAM, 64k CCM RAM, 1024k Flash
HARDWARE: STM32F407VGT6 168MHz, 128KB RAM, 1MB Flash
DEBUG: Current (jlink) External (jlink, stlink)
PACKAGES:
 - framework-arduinoststm32 0.0.0
 - framework-cmsis 2.50501.200527 (5.5.1)
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Library Manager: Installing bitbucket-fmalpartida/LiquidCrystal @ 1.5.0
KeyError: 'framework-stm32cube':
  File "C:\Users\Anthony\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 181:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\Anthony\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Script\SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Anthony\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Script\SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\Anthony\.platformio\platforms\ststm32@10.0.1\builder\main.py", line 99:
    target_elf = env.BuildProgram()
  File "C:\Users\Anthony\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Util.py", line 742:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Anthony\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 62:
    env.ProcessProjectDeps()
  File "C:\Users\Anthony\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Util.py", line 742:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Anthony\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 141:
    project_lib_builder = env.ConfigureProjectLibBuilder()
  File "C:\Users\Anthony\.platformio\packages\tool-scons\scons-local-4.3.0\SCons\Util.py", line 742:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Anthony\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 1104:
    project.install_dependencies()
  File "C:\Users\Anthony\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 940:
    lm.install(spec)
  File "C:\Users\Anthony\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 48:
    pkg = self._install(
  File "C:\Users\Anthony\.platformio\penv\lib\site-packages\platformio\package\manager\library.py", line 105:
    if is_builtin_lib(spec.name):
  File "C:\Users\Anthony\.platformio\penv\lib\site-packages\platformio\commands\lib\helpers.py", line 49:
    for storage in storages or get_builtin_libs():
  File "C:\Users\Anthony\.platformio\penv\lib\site-packages\platformio\commands\lib\helpers.py", line 34:
    for storage in p.get_lib_storages():
  File "C:\Users\Anthony\.platformio\penv\lib\site-packages\platformio\platform\base.py", line 214:
    pkg = self.get_package(opts["package"])
  File "C:\Users\Anthony\.platformio\penv\lib\site-packages\platformio\platform\_packages.py", line 30:
    return self.pm.get_package(spec or self.get_package_spec(name))
  File "C:\Users\Anthony\.platformio\penv\lib\site-packages\platformio\platform\_packages.py", line 22:
    owner=self.packages[name].get("owner"),
=============================================================================================== [FAILED] Took 2.08 seconds ===============================================================================================

Environment                        Status    Duration
---------------------------------  --------  ------------
mks_robin_nano_v3_usb_flash_drive  FAILED    00:00:02.081
========================================================================================= 1 failed, 0 succeeded in 00:00:02.081 ========================================================================================= 
E:\ADragone\OneDrive\Computers\Nomad\Desktop\Marlin-SuperRacer-MKS-Nano-V3-26-02-2022> echo "done" >"C:\Users\Anthony\AppData\Local\Temp\ipc"
```

antdragone, 2022-03-09:
With the help of the Marlin Devs in their discord, we were able to compile the attached folder, however it doesn't work properly on the machine itself.
[Marlin-2.0.9.3.zip](https://github.com/Guilouz/Marlin-SuperRacer-MKS-Nano-V3/files/8214514/Marlin-2.0.9.3.zip)

antdragone, 2022-03-09:
So, I went ahead and tried to do this from scratch.

I downloaded the repo clean, and attempted to compile, and it failed (as mentioned it would)
I then copied the file MarlinSerial.ccp.o file to the \.pio\ folder, in the requested destination.
And then I attempted to build again (I have not changed any files at this point) and it failed again.

I am getting the same set of errors every time I compile. I feel like something might be wrong with my machine, but I'm able to compile firmware for my Ender 3 and Ender 2 without issue. So I'm really scratching my head on this one.

ellensp, 2022-03-09:
The "KeyError: 'framework-stm32cube':" is caused by installing the wrong Liquid crystal library

platformio.ini has this line  

`USES_LIQUIDCRYSTAL      = bitbucket-fmalpartida/LiquidCrystal@1.5.0`

But this is not the correct library
It should be using  bitbucket.org/fmalpartida/new-liquidcrystal but this no github longer exists

Lucky marlin took a fork so  we can use that, by changing that line to
`USES_LIQUIDCRYSTAL      = LiquidCrystal=https://github.com/MarlinFirmware/New-LiquidCrystal/archive/1.5.1.zip`

ellensp, 2022-03-09:
Found a second issue (just documenting for others) 
antdragone was using Auto Build Marlin plugin. this wipes .pio directory every time it is run.

antdragone, 2022-03-09:
Yeah - My bad! But the `USES_LIQUIDCRYSTAL = LiquidCrystal=https://github.com/MarlinFirmware/New-LiquidCrystal/archive/1.5.1.zip` Resolved our first issue.

ellensp, 2022-03-09:
@Guilouz Can you confirm that this replacement MarlinSerial.cpp.o works under windows too?  it works myself under linux, but antdragone is a windows user.
Is now getting the error

```
c:/users/anthony/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe:.pio/build/mks_robin_nano_v3_usb_flash_drive_msc/src/src/HAL/STM32/MarlinSerial.cpp.o: file format not recognized; treating as linker script
```

antdragone, 2022-03-09:
@ellensp was able to compile without issue using windows.

I am now getting this current error, after updating the LiquidCrystal line in Platformio.ini and after following the instructions to replace MarlinSerial.cpp.o (after letting it build, fail, and then replace, and build).

```
c:/users/anthony/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe:.pio/build/mks_robin_nano_v3_usb_flash_drive_msc/src/src/HAL/STM32/MarlinSerial.cpp.o: file format not recognized; treating as linker script     
c:/users/anthony/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe:.pio/build/mks_robin_nano_v3_usb_flash_drive_msc/src/src/HAL/STM32/MarlinSerial.cpp.o:7: syntax error
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\mks_robin_nano_v3_usb_flash_drive_msc\firmware.elf] Error 1
```

Guilouz, 2022-03-09:
I’m on macOS.
What changes exactly do you need in configuration.h/configuration_adv.h ?

antdragone, 2022-03-09:
Yeah, I'm thinking this is an issue with my machine.

I simply need the following changes made;
```
#define TEMP_SENSOR_0 5
#define HEATER_0_MAXTEMP 300
#define DEFAULT_AXIS_STEPS_PER_UNIT   { 80, 80, 80, 400 }
```
Thank you for your help, this has been a head scratcher for me.

Guilouz, 2022-03-09:
[image]

SDCARD : 
[Robin_nano_v3.bin.zip](https://github.com/Guilouz/Marlin-SuperRacer-MKS-Nano-V3/files/8216191/Robin_nano_v3.bin.zip)

USB : 
[Robin_nano_v3.bin.zip](https://github.com/Guilouz/Marlin-SuperRacer-MKS-Nano-V3/files/8216209/Robin_nano_v3.bin.zip)

antdragone, 2022-03-09:
Huh, now i'm getting the Error TH-MAX, and it won't boot while I have the thermistor plugged in.

antdragone, 2022-03-09:
It was working with a previous thermistor, so It may just be this new one, I'll take apart the solder job and investigate. I'll keep you posted. This part is probably a hardware problem.

antdragone, 2022-03-09:
Can confirm the issue was with a short in the wires, but now the machine isn't showing any data, and isn't responding to my inputs....
[image]

antdragone, 2022-03-09:
> Can confirm the issue was with a short in the wires, but now the machine isn't showing any data, and isn't responding to my inputs.... [image]

Reflashing the firmware fixed this

antdragone, 2022-03-09:
> [image]
> 
> SDCARD : [Robin_nano_v3.bin.zip](https://github.com/Guilouz/Marlin-SuperRacer-MKS-Nano-V3/files/8216191/Robin_nano_v3.bin.zip)
> 
> USB : [Robin_nano_v3.bin.zip](https://github.com/Guilouz/Marlin-SuperRacer-MKS-Nano-V3/files/8216209/Robin_nano_v3.bin.zip)

Thank you! With your help I'm going to be able to use the E3D Revo on the SR! I sincerely appreciate the work you do!

I'll be sure to leave a donation once I get paid 😅

closed 2022-03-09 by antdragone

Guilouz, 2022-03-09:
Good thing ! Perfect !