Guilouz/Marlin-SuperRacer-MKS-Nano-V3 #23: question about firmware

aleZZ2001, 2022-03-09:
hello, i have two question for you.

the screen update is for the stock screen?

the bondtech lgx bowden is the bondtech bmg? https://www.bondtech.se/product/bmg-extruder/

Guilouz, 2022-03-09:
Yes screen update is for stock screen, it's the same file v1.3 as flsun website.

No Bondtech LGX is this one : https://www.bondtech.se/product/lgx-large-gears-extruder/

For BMG, no need to change settings, it's the same as stock one.

aleZZ2001, 2022-03-09:
ok, thanks, today i received the new extruder and probably i'm gonna change firmware

closed 2022-03-09 by Guilouz

aleZZ2001, 2022-03-10:
hello again, did you have a good cura or any other slicer profile to share? i tried a lot of profiles for any slicer but all are bad