Guilouz/Marlin-SuperRacer-MKS-Nano-V3 #27: Nach Firmware Installation von LGX Lite Version kommt Fehlermeldung "Kein Drucker" und startet ständig neu

MichaelLehner69, 2022-04-25:
Hallo Cyril,

ich hab ein TFT70 GD am Nano V3 und es funktioniert alles. Habe heute den LGX Lite montiert und deine Firmware 
"SR-LGX-Lite-11-03-2022-SDCARD" verwendet.  Danach startet der Drucker ständig neu und sagt, dass kein Drucker angeschlossen ist. Hab jetzt wieder deine aktuelle "TFT70-22-04-2022-GD" installiert und er funktioniert wieder. Was könnte der Fehler sein? Oder kann ich über gcodes die nötigen Einstellungen auch mit dieser Firmware eingeben? Vielen Dank schonmal im Voraus... und schick mir bitte einen paypal Link, ich überweise dir 20 Euro. liebe Grüße

MichaelLehner69, 2022-04-25:
ich hab folgende Befehle ausgeführt, aber der Extruder läuft leider in die falsche Richtung

M92 E562
M906 E650
M603 L70.0000 U70.0000  
M500
M501

MichaelLehner69, 2022-04-25:
Hi Cyril, i solved the problem. myself I just used the wrong file, now i took  "Marlin 2.0.9.3 - FLSUN SR - BTT TFT - NanoV3 - LGX Lite" from the Pack Super Racer TFT70 which i overlooked... Thanks for your hard work and the stl's to mount the LGX Lite... take a look at your paypal account and enjoy  :)

Guilouz, 2022-04-25:
All is ok for you ? Thanks for your donation :)

MichaelLehner69, 2022-04-25:
yes, now everything is fine 👍

closed 2022-04-28 by Guilouz