Guilouz/Marlin-SuperRacer-MKS-Nano-V3 #29: version for mks ts_35

smiley-2, 2022-04-29:
Bonjour,

serai-t'il possible d'avoir une version de ton firmware pour les btt- tft70  adapté pour les mks ts-35? 

j'ai essayé de modifier tes sources, mais je ne suis pas encore assez bon pour avoir un truc réussi.

merci!

Guilouz, 2022-04-29:
MKS TS 35 ou MKS TFT 35 ?

Voilà la liste des MKS compatibles avec le firmware BTT : 
;MKS_TFT28_V3_0
;MKS_TFT28_V4_0
;MKS_TFT28_NEW_GENIUS
;MKS_TFT32_V1_3
;MKS_TFT32_V1_4
;MKS_TFT32_V1_4_NOBL
;MKS_TFT35_V1_0

smiley-2, 2022-04-29:
merci  pour ta réponse, mais c'est bien le mks ts 35
https://fr.aliexpress.com/item/1005001446396197.html?spm=a2g0o.order_list.0.0.6cbb5e5bS0Vwl4&gatewayAdapt=glo2fra

trop facile sinon!

je sais qu'il faut activer tft_color ui j'ai réussi à avoir un affichage mais avec des bugs, j'ai essayé d'autre firmware alternatifs qui fonctionnaient, mais j'ai une préférence pour les tiens.

de plus tu penses faire un mise à jour à partir du firmware 1.4 de flsun? (pour des proches qui n'ont pas changé d'écran)

merci!

Guilouz, 2022-04-29:
Je n'ai aucune idée si le fw TFT35 fonctionne sur le TS35, tu peux tester pour voir.

Quand flsun aura décidé de partager les sources, c'est normalement prévu à la sortie de la V400.

smiley-2, 2022-04-29:
Franchement, je pense avoir déjà testé, ( mais je vieillis et ça fait
quelques mois maintenant ) je vais réessayer et te dis cela. Mais je ne
crois pas que ça fonctionne.
Mon souci en recompilant tes sources c'est que je n'arrivais pas à enlever
ce que tu avais ajouter pour le tft70  afin de rajouter mes deux lignes de
modifs

Mais merci encore pour tes réponses et ton taf, tu passes pas en Alsace un
jour ? Plutôt que payer une bière via PayPal je préfère en offrir une vraie
d'un brasseur artisanal du coin!

smiley-2, 2022-05-01:
je confirme, ça ne fonctionne pas, en même c'est logique car on ne peut pas flasher de firmware sur le ts35!

Guilouz, 2022-05-01:
Ah oui il doit utiliser l'UI dans Marlin alors. Donc tu peux oublier le firmware BTT dessus.

closed 2022-05-01 by Guilouz

smiley-2, 2022-10-11:
> <Guilouz/Marlin-SuperRacer-MKS-Nano-V3/issue/29/issue_event/6530082113@
> github.com>

C'est ça, mais ne peux-tu pas faire une version de ton firmware avec l ui
marlin? Ça serait possible stp?

Guilouz, 2022-10-11:
Non désolé Marlin c'est terminé, place à Klipper dorénavant.

smiley-2, 2022-10-12:
pas de problème j'y suis passé aussi et c'est carrément mieux! encore merci
pour tout ton boulot!