Guilouz/Marlin-SuperRacer-MKS-Nano-V3 #31: reboot

RoadArk, 2022-07-16:
After swap new motherboard (3.0 > mks robin nano 3.1) my printer restart in full print in a completely random way without displaying a message, it reboots purely and simply .... I already had the problem with version 3.0 but I also have it with version 3.1

screen version : v1.3
motherboard firmware : lgx lite only sdcard version 2.0.8

Guilouz, 2022-07-16:
Firmware is for MKS Robin Nano V3.0 not for V3.1 which does not have the same CPU (STM32F407VET6 for 3.1 and STM32F407VGT6 for 3.0) and not the same flash size (512KB Flash for 3.1 and 1024KB Flash for 3.0).

RoadArk, 2022-07-16:
go !  work !  hop hop hop... XD

RoadArk, 2022-07-16:
otherwise excluding the unpredictable frizz, it works well!  I changed SD card to see if there is a better one.  I launched a print of 2h30 no worries

RoadArk, 2022-08-07:
after change microSD this issue is solved ! I confirm, the firmware it's fully compatible on the MKS Robin Nano V3.1 motherboard ! GREAT :)

closed 2022-08-07 by RoadArk