Guilouz/Marlin-SuperRacer-MKS-Nano-V3 #32: I have a problem FLSUN has released the firmware 1.4 and an update for the monitor.

Gueni49, 2022-07-16:
Hi

I like your work, only I have a problem FLSUN has released the firmware 1.4 and an update for the monitor.
You can now enter more. However, you do not have the Sours code to download.

I'm not that much of a programmer but I can set the default stuff. 
I have installed a mosquito and have to change the temp sensor to the number 5 and the pressure temperature to 450 degrees.
 Your version does not fully support the display ( firmware 1.4 ) at the moment, I would be grateful if you could adjust this on occasion.

Greetings Michael

tucznak, 2022-09-25:
Seconded. It would be great to get an update for this, maybe move up to a higher version of Marlin. Or even better yet, outright merge the necessary fixes for SR to mainline Marlin, so we wouldn't have to maintain a separate outdated branch...

Guilouz, 2022-09-25:
@tucznak without screen source code it’s not possible. And same for v1.4 without their marlin source code nothing can be done.

closed 2022-09-25 by Guilouz

Guilouz, 2022-09-26:
Yes you can send me.

mattia480, 2023-02-22:
> @Guilouz I actually have the source for 1.4 Nano, if you want? Let me know of a good way to pass them to you.

@Guilouz, i'm currently working on a marlin 2.1 version for SR with original display at version 1.4 and MKS Robin Wifi support. I'm still blocked because i don't have the FLSun 1.4 firmware source code. Can you send me the file so I can complete the project and post it on GitHub?
I leader here my email address: mattia480@gmail.com 
Thank you!

Guilouz, 2023-02-22:
Source code for 1.4 is not available.

tucznak, 2023-02-22:
Source code for 1.4 is available. They don't have it on their website, but since it's GPL they have to give it to you if you ask for it, which I did back in September. I was hoping Guilouz would give me a way to send him the files... Anyway, I've now uploaded them to some hosting service, you can download it from here: https://ufile.io/hp5w9eiu
It's still 2.0.8 bugfix with minimal changes from 1.3 it seems, but I didn't dig too much into it.

mattia480, 2023-02-22:
> Source code for 1.4 is available. They don't have it on their website, but since it's GPL they have to give it to you if you ask for it, which I did back in September. I was hoping Guilouz would [...]

Thank you so much!!! I've asked to the FLSun support email, but this was the answer:
From: service <service@flsun3d.com>
Date: 21/02/23 01:48 (GMT+01:00)
To: Mattia
Subject: Re:Super Racer WiFi Firmware

Hello friend. I apologize. We can't provide you with 1.4 open source firmware for the time being

发件人：Mattia
发送日期：2023-02-20 18:27:52
收件人："service@flsun3d.com" <service@flsun3d.com>
主题：Super Racer WiFi Firmware
Hello, i've an FLSun Super Racer with stock MKS Robin Nano V3.1 and i've added the MKS Robin WiFi module 1.1 with ESP3D firmware.
The wifi interface does't connect to the printer, i think beacuse the wifi module is not enable in Marlin firmware 1.4, can you provide me firmware 1.4 with wifi support enabled or a link to firmware 1.4 SDK so i can compile it?
On your [website ](https://flsun3d.com/pages/sr-firmware)i can only find th [SDK ](https://drive.google.com/file/d/1HGFdXoqVZvl5m7ymgPkb2Bxv0XByutFy/view?usp=sharing) for 1.3 firmware?
Thank you,

Mattia

tucznak, 2023-02-22:
In which case you should remind them about the terms of GPLv3 that Marlin is licensed under and that they've just tried to break with that e-mail ;-) I see now on their web that they still haven't uploaded the 1.4 files, and that they're also not providing Klipper sources at all (also under GPLv3). But we're getting offtopic here, glad I could be of some help, I'll look into the license issues somewhere where it's more appropriate, not in a github issue :-)