Guilouz/Marlin-SuperRacer-MKS-Nano-V3 #33: Problems with the current version

GameboysLP, 2022-09-15:
- Time for printed time is incorrect
- When printing 100mm, it doesn't print 100mm, it's only 5mm
- Eprom and console are without text
- Eprom is empty when clicked
- Temperature is not properly regulated
- Printer simply cancels printing

Guilouz, 2022-09-15:
If you have console i think you have screen firmware 1.4.
This firmware can only be used with version 1.3 of the screen.

GameboysLP, 2022-09-15:
Hey, I have screen version 1.3

harrisck, 2022-09-27:
> Hey, I have screen version 1.3

if you had v1.3 installed onto the screen you would not see the eeprom button, that was only added in 1.4. reflash the screen with v1.3 from the flsun downloads page and it will work

itajackass, 2023-08-24:
Same problem for me (no text for eeprom and console). I came from original 1.4...and of course i've flashed 1.3 onto screen...
Also... Z offset adjust (during printing adjustment) show me only +/- 0.05...... before there was also +/- 0.00125

Any suggest?

Kadgey, 2023-08-24:
> Same problem for me (no text for eeprom and console). I came from original 1.4...and of course i've flashed 1.3 onto screen... Also... Z offset adjust (during printing adjustment) show me only +/- 0.05...... before there was also +/- 0.00125
> 
> Any suggest?

I would flash 1.3 on the screen again   because you shouldn't be seeing the eeprom and console option as they didn't exist in V1.3 them options was an addition to V1.4. Had same issue myself the second flash fixed it. 👍